robotology/mesh-support #2088: `r_index` broken

mebbaid, 2025-03-26:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

During a teleoperation session, the hand crashed against the black table breaking the plastic connection of the finger first phalang to the rotating rod. 

### Detailed context

[image]

### How does it affect you?

This affects a demo planned for tomorrow and the day after tomorrow.

Fabrizio69, 2025-03-27:
@fbiggi is working on this issue....

fbiggi, 2025-03-27:
Operations performed:

- [ ] Dismantling of the finger from the hand

- [ ] Realignment of the brass connecting rod

- [ ] Replacement of the plastic part and reworking with boring and tapping

- [ ] Reassembly of the finger on the hand

[image]

[image]

[image]

[image]

S-Dafarra, 2025-03-27:
Awesome, thanks a lot!

closed 2025-03-31 by AntonioConsilvio