robotology/mesh-support #2089: Lidar does not transmit data

vigisushrutha23, 2025-03-27:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

Similar to Issue #2052 we get the follwing error when we run yarprobotinterdate --config sensors.xml on the ergocub-head. 

[image]

### Detailed context

No Lidar data is transmitted as there is either a cable fault or an issue with transmission to the head computer. 

### How does it affect you?

This blocks us from doing any tests/demos involving navigation.

SimoneMic, 2025-03-28:
I have tested the lidar and it works, I am able to read data from the port

vigisushrutha23, 2025-03-31:
@SimoneMic ran the following fixes for the SN001.

- Update yarp on the head to 3.10.1
- Downgrade yarp on robot laptop to 3.10.1
- Update CMake to 3.20 on the head.

Following these changes, the Lidar data was properly transmitted and could be used with the navigation modules. So we are closing this issue.

closed 2025-03-31 by vigisushrutha23

AntonioConsilvio, 2025-03-31:
Thank you for the explanation!