robotology/mesh-support #2092: right_leg-eb6-j0_3 Motor temperature detection board is not receiving any data on the i2C channel

SimoneMic, 2025-03-28:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

After startup the right hip pitch goes in hardware fault with the following message:

```
[ERROR] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=774s 758m 425u :  MC: TDB cannot receive any data on I2C channel. Motor temperature detection board is not receiving any data on the i2C channel. (Joint=r_hip_pitch (NIB=0))

```

### Detailed context

No actions have been performed before this issue and the error presents itself even after startup

SimoneMic, 2025-03-28:
After a reboot, I  have this same error, but on the left one

Fabrizio69, 2025-03-28:
With @Gandoo checked the wiring for right hip pitch (from RL_TDB0 to 2FOC-6B0), it's OK...

[image]

cc: @AntonioConsilvio @MSECode @valegagge

mebbaid, 2025-03-28:
Just tried to use the robot, and I am running into the same issue.

```bash
[ERROR] from BOARD 10.0.1.6 (right_leg-eb6-j0_3) time=489s 703m 675u :  MC: TDB cannot receive any data on I2C channel. Motor temperature detection board is not receiving any data on the i2C channel. (Joint=r_hip_pitch (NIB=0))

```

[image]

MSECode, 2025-03-28:
Hi, 
Since that is a joint that might have some issues due to the fact that the analog cable section is quite long, it might generate that error in some conditions. 
Therefore, I suggest you to disable the temperature sensor for now setting it to `NONE` for that joint. 
Thanks