robotology/mesh-support #2095: Arms shoulder motors might have joints backlash or slippage - investigation needed

HosameldinMohamed, 2025-03-31:
### Robot Name 🤖

iCubGenova09 (iRonCub3) S/N:000

### Request/Failure description

During iRonCub flight experiments, we noticed that the shoulder joints (roll, pitch and/or yaw) are recently having some slippage or backlash.

Probably in investigation is needed to understand the condition of the motors/reducers

### Detailed context

The way we found out this is that after many tests we set the robot joints to home position and some arms joints looked distorted, when we restart the robot and perform type 10 calibration the robot status is restored.

This issue is become more frequent with more tests we perform

### How does it affect you?

We have the last week of iRonCub experiments in iit! and this issue is affecting the flight tests.

CC @Giulero @vpunithreddy @pillai-s @FabioBergonti @davidegorbani @STaliani @xela-95 @DanielePucci

maggia80, 2025-03-31:
@HosameldinMohamed, the joints have a clutch. If you have logged both motor and joint position, you could see if there is a drift between the two measurements. This is true in case you are using the AMO encoder for the joint position.

fbiggi, 2025-03-31:
Checking the screws and register:

- [x]  4pz V4-10--_-_U5931_GC.
- [x]  6pz V4-10--_-_U5739
- [x]  8pz V4-8--_-_U5931_GC
- [x]  2pz ICUB3_001_P_279
- [x]  2pz G3-6--_-_U5923_G
The marked screws were loose, after calibration and everything seems ok
Waiting for feedback after flight test

[image]

@maggia80 @Fabrizio69 @AntonioConsilvio

closed 2025-12-24 by pattacini