robotology/mesh-support #2098: Right Hand adduction/abduction problem

Ep3896, 2025-03-31:
### Robot Name 🤖

iCubErzelli02 S/N:036

### Request/Failure description

Following today's intervention, an issue was identified with the right-hand: the adduction/abduction movements are not being executed, as joint 7 remains unresponsive.

### Detailed context

cc @ddetommaso 

### Additional context

[log_icub-node_yarprobotinterface_2015.txt](https://github.com/user-attachments/files/19534938/log_icub-node_yarprobotinterface_2015.txt)

[image]

https://github.com/user-attachments/assets/fed678d4-2958-48f4-8c2f-bfc0d08d2117

### How does it affect you?

This problem is of high priority, though not critical; however, it must be resolved to proceed with the experiments.

AntonioConsilvio, 2025-03-31:
Hi @Ep3896 @ddetommaso! The finger adduction tendon is probably broken. 
Replacing that tendon takes a few days between electronics, mechanics and testing.

Also to work better we should have the lowerarm detached from the robot. 

So let us know when the robot is not going to be used and it can stay without the right lowerarm for a few days, so we can bring it to our lab and put it back on the robot when we finished.

ddetommaso, 2025-04-03:
Hi @AntonioConsilvio ,

thank you for your quick feedback. We agreed internally that from next Tue we can stay without the robot during the maintenance period (possibly not more than one week). Please let me know if this plan fits with your timeline.

Thank you

ddetommaso, 2025-04-14:
Hi @AntonioConsilvio,

can you please update us about the status of the maintenance?
When u think would you guys be available to replace the arm?

Thank you

cc @Ep3896

AntonioConsilvio, 2025-04-15:
HI @ddetommaso, we are in the testing phase, you will receive updates soon!

filippoborgogni, 2025-04-15:
Hi @ddetommaso , the maintenance work is finished; tomorrow @edersonferrante and I will come to reassemble the arm on the robot!
If possible it could be very helpful align `Robots-Configuration` to this [commit](https://github.com/robotology/robots-configuration/pull/726).
Thanks!

ddetommaso, 2025-04-15:
Hi @AntonioConsilvio 

unfortunately we didn't have time to align the configuration on the robot today. I would please ask you guys to do it tomorrow. Thank you

cc @Ep3896

Fabrizio69, 2025-04-30:
Hi @Ep3896, as @filippoborgogni pointed out in the [previous comment](https://github.com/robotology/icub-tech-support/issues/2098#issuecomment-2812150242), you have to align the robot configuration to the approved PR and close this issue about the right hand adduction.

For the problem encountered today: a finger remains bent, you need to open a new issue. Thanks

cc: @AntonioConsilvio