robotology/mesh-support #2099: Torso pitch goes in over-current protection at startup

SimoneMic, 2025-03-31:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

When starting the robot, the torso pitch goes in over-current protection as you can see from this picture:

[image]

When restarting the singular motor from the motorgui, it still goes in the same error. It's not possible to recover

### Detailed context

Probably related to #1981 or #2077

Gandoo, 2025-05-14:
ciao, WM1 torso engine is replaced, we have to sectional the Lower from the Upper for this replacement

cc @AntonioConsilvio @SimoneMic @isorrentino

filippoborgogni, 2025-05-19:
Hi @SimoneMic,
today with @AntonioConsilvio we made the rotor-stator autocalibration for the joint of the `torso pitch` , we calibrated the AMO sensor and we tested it.
During the testing phase the cable (label WE1) that connects the 2foc board (label 5B1) to the LCORE of the `torso pitch` broke and I fixed it.
Now the robot works properly and the joint is able to move in all of its `range of motion` as can be seen in the video below! 

https://github.com/user-attachments/assets/43c5da7c-ed78-4328-855c-5f98fc29ac4b

I proceed to close this issue and the [linked issue](https://github.com/robotology/icub-tech-support/issues/2077
)! 🚀

cc @Fabrizio69

closed 2025-05-19 by filippoborgogni