robotology/mesh-support #2100: ergoCub 1.1 S/N:001 – l_middle_oc not working

PasMarra, 2025-04-01:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

as per title, `l_middle_oc`does not work.

### Detailed context

As an example, here I'm setting a reference of `90 deg`,  but the finger does not move at all.

[image]

filippoborgogni, 2025-04-02:
Hi @PasMarra , 
after testing, I could see that the joint is working properly, but no work has been done on either the hardware or software, so why it had previously stopped working must be investigated.

cc @Fabrizio69 @AntonioConsilvio

AntonioConsilvio, 2025-05-05:
HI @PasMarra, do you know if this problem happened again?

PasMarra, 2025-05-05:
Hi @AntonioConsilvio, i haven't used this robot in a while, so I don't know its current state.