robotology/mesh-support #2102: Right anklle roll and pitch encoders jump

GiulioRomualdi, 2025-04-02:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

We started the robot and the right ankle pitch suddenly jumped. We checked the encoder value and even if the joint was manually p[aced in zero. The values where highly different from zero

### Detailed context

We cannot use the AMO on the ankle

GiulioRomualdi, 2025-04-02:
cc @fbiggi @filippoborgogni

filippoborgogni, 2025-04-09:
Hi @GiulioRomualdi , I fixed the cable that connects the AMO board of the ankle pitch to the EB7 board (ems4). 
I tested the robot doing a cycle of movement and both joints (ankle pitch and ankle roll) seems to work properly. 
Give us a feedback ! 🚀

cc @AntonioConsilvio

GiulioRomualdi, 2025-04-09:
Hi @filippoborgogni, I tested the robot. The right ankle pitch remains uncalibrated. Moving the joint manually, the encoder value doesn't change

filippoborgogni, 2025-05-15:
Hi @GiulioRomualdi ,
yesterday with @edersonferrante I worked to solve this issue.
I could see, during the debugging phase, that the cable (label RLS4) connecting the ems4 board (label EB7) to the AMO board of the `right ankle pitch` joint was damaged (in the photo below you can see that the red cable was broken). 

[image]

I fixed the cable and checked that the robot was working properly. During this phase I could also see that the AMO board reading of the `right knee` joint was wrong; I then observed how the cable (label RLS3) had the red wire coming out of the connector and how the connector was damaged. 

[image]

I then replaced the connector and reconnected the sensor to the ems4 board (label EB6).
I then checked that the robot worked correctly and that the joints of the `right ankle pitch` and `right ankle roll` were able to move throughout the `range of motion` (test which can be seen in the video below).

https://github.com/user-attachments/assets/67230a5a-83dd-4939-91c3-ac53f14213e9

I also made a [commit ](https://github.com/icub-tech-iit/robots-configuration/commit/e0674a4a797343956e38183bd83b417fbda1c2c4)in `robots-configuration` to enable again the AMO reading for the `ankle pitch` joint (the file that i changed is `hardware/motorControl/right_leg-eb7-j4_5-mc_service.xml`)

Closing!🚀

cc @AntonioConsilvio @Fabrizio69

closed 2025-05-15 by filippoborgogni