robotology/mesh-support #2103: [Provide CAN access to hand skin boards]

simeonedussoni, 2025-04-03:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

In view of a grasping/handling demo dedicated to log also skin data, I need to be sure that both hands are capable of sending the skin data. since yarprobotinterface sometimes get stuck, the best option is to connect a CAN cable directly to the skin MTB4 card on each hand.

### Detailed context

on each hand I need t have a CAN splitter connected to the skin MTB4 board with the CAN cable accessible from outside. In this way if the skin get stuck I can send reset commands without relaunch of the whole yarprobotinterface and eventually have the possibility to vary some parameters to have a clearer picture of the expected signal. The demo is foreseen for **wednesday, april 9 from 10 to 13 **

thank you very much in advance

Fabrizio69, 2025-04-04:
Hi @simeonedussoni this is not a manintenance request. This is a change request and I understand it's the same thing [requested here and already completed](https://github.com/icub-tech-iit/tickets/issues/3858).
Regarding the skin, do you have [this pending problem](https://github.com/icub-tech-iit/proto/issues/3195#issuecomment-2725210588)?

simeonedussoni, 2025-04-09:
ciao @Fabrizio69 
thanks for highlighting me the connected issue. 
it seems is the feature I needed. We can close this issue. 
cc @Gandoo @AntonioConsilvio @aazocar7

closed 2025-04-09 by simeonedussoni