robotology/mesh-support #2104: R1 S/N:003 error on elbow and fingers when starting robotinterface

ste93, 2025-04-04:
### Robot Name 🤖

R1 S/N:003

### Request/Failure description

R1 gives the error below when starting the robotinterface, It seems a problem of dirty encoder for the left elbow and some fingers need re-calibration as discussed yesterday with @MSECode.

### Detailed context

when starting the robot interface, as soon as the robot calibrates the yaw, the elbow goes idle. in addition, the fingers sometimes fails or goes idle during the calibration

### Additional context

[image]
[image]

filippoborgogni, 2025-04-04:
Hi @ste93,
Today we have been working on solving this problem, which we have observed only occurs during the calibration phase, as by running the elbow from the `yarpmotorgui` once the calibration is complete the joint moves correctly.
This morning together with @fbiggi we cleaned the optical encoder (as you can see in the photo it was quite dirty) and the error message disappeared, although the problem had not disappeared.

[image]351d)

During the afternoon with @aazocar7 we worked on the wiring to try to debug the problem and, after various tests and trials, we could see that all the cables were continuous and none were visibly damaged.
We then tried to reproduce the situation where the elbow goes into idle with the movement of the shoulder yaw joint by creating a loop that would reproduce the movement that was causing problems during calibration (as seen in the video below) but everything worked perfectly. At last we could also see how the error message of the dirty optical encoder (which had previously disappeared) reappeared despite the reading being correct.

https://github.com/user-attachments/assets/ef489b53-f374-4fb3-9cae-c32e134e13fe

One last note: we were able to observe how some shoulder joint encoders (in the photo you can see the shoulder pitch of the left arm) have a faulty reading with a situation where the actual position of the joint does not match the encoder reading.

![Image](https://github.com/user-attachments/assets/00c94dee-ba26-470f-aab2-8

2691f9378ac)

cc @AntonioConsilvio @Fabrizio69

filippoborgogni, 2025-04-08:
Hi @ste93 , with @AntonioConsilvio we have been working over the last few days to solve these problems.

After several tests, we could see that the left elbow joint had much more friction than the right one. 
We therefore decided to run the joint by creating a movement loop and making it move all day. Despite this, the friction remained more pronounced and during the calibration phase it continued to idle and did not move in the expected way, as the joint could not reach the setpoint. 
We therefore modified the configuration files by increasing the PWM values and the maximum current that the elbow could use to reach the setpoint. After testing several times we could see that the problem started to occur a lot less (in some rare cases it still happens, you have to keep using the joint to decrease the friction of the motor gearbox) and we can therefore consider the problem solved. 

This is the [PR link](https://github.com/robotology/robots-configuration/pull/724), please align the robot's `robots-configuration` when possible!
We have also solved the thumb oc problem by replacing the FAP cable which had stopped working. 
As far as other fingers are concerned, the problem has not reoccurred, if it does, please open a new dedicated issue.

One note: during the testing and debugging phase of the problem, the EB31 (amc) board started to timeout, and we later realised that the problem was related to the fact that the CAN loop connecting the board to the 2 amcbldc boards was damaged; this problem is now solved too! 

Closing!🚀

closed 2025-04-08 by filippoborgogni

ste93, 2025-04-08:
@elandini84 fyi 
Thanks @filippoborgogni