robotology/mesh-support #2105: l_middle_oc does not move

PasMarra, 2025-04-04:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

`l_middle_oc` does not move.

### Detailed context

In blue the reference and in red the measured value for the `l_middle_oc` . As you can see, the finger is not moving at all. 

[image]

edersonferrante, 2025-04-07:
Hi @PasMarra , as you can see from the video the problem is due to the fact that the tendon terminal has slipped out of the pulley, this is caused by the hole in the pulley which might be narrow compared to the diameter of the terminal so it would tend to come out upward.

https://github.com/user-attachments/assets/f5cfb053-d1a3-4693-b43e-583d4f780de9

Me and @AntonioConsilvio did a test with the middle finger of the right hand and the terminal for tendons doesn't come out because the hole of the pulley is wider so the terminal for tendons doesn't tend to come out upwards but downwards ( I noticed that the harstop Cod. 16399 Rev. A1 moves so as soon as it's possible we'll screw its screw back in and put the threadlocker ).

https://github.com/user-attachments/assets/30924bcf-0193-4a94-acdc-2baf611ba57d

[image]

In this photo below, I first took a picture of how I found the terminal for tendons, then I put the terminal for tendons back inside its pulley, and lastly I tentatively put American tape ( Code 15441) to prevent the Terminal for tendons from coming out of its seat.

[image]

cc @Fabrizio69 @maggia80

AntonioConsilvio, 2025-04-08:
Just for info, the first time the problem occurred was in this issue:

- https://github.com/robotology/icub-tech-support/issues/2074

ps. @edersonferrante also fixed this problem:

> ( I noticed that the harstop Cod. 16399 Rev. A1 moves so as soon as it's possible we'll screw its screw back in and put the threadlocker ).