robotology/mesh-support #2107: Linear testbench fault button cable broken

isorrentino, 2025-04-07:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

The cable of the fault button of the linear testbench is broken.

### Detailed context

[image]

cc @FabioBergonti

isorrentino, 2025-04-07:
I manually changed the robot name in the first comment as the linear-actuator-testbench is not in the list.

cc @AntonioConsilvio @Fabrizio69

pattacini, 2025-04-07:
Hi @isorrentino 

Can you please rather rely on our internal ticketing system for that?

isorrentino, 2025-04-07:
Thank you @pattacini