robotology/mesh-support #2111: [Provide a meaningful TITLE, but DON'T enter the robot name here]

valegagge, 2025-04-08:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

the yarprobotinterface doesn't start due to this problem:

```
[ERROR] theNVmanager::Impl::wait() had a timeout for BOARD left_arm-eb31-j4_6 IP 10.0.1.31 and nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
[ERROR] theNVmanager::Impl::command() fails a wait() from IP 10.0.1.31 for nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
[ERROR] theNVmanager::Impl::wait() had a timeout for BOARD left_arm-eb31-j4_6 IP 10.0.1.31 and nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
[ERROR] theNVmanager::Impl::command() fails a wait() from IP 10.0.1.31 for nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
[ERROR] theNVmanager::Impl::wait() had a timeout for BOARD left_arm-eb31-j4_6 IP 10.0.1.31 and nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
[ERROR] theNVmanager::Impl::command() fails a wait() from IP 10.0.1.31 for nv ID32 = 0x00030001 -> IND = 0, TAG = eoprot_tag_mn_service_status_commandresult
[ERROR] EthResource::serviceCommand() failed an acked activation request to BOARD left_arm-eb31-j4_6 with IP 10.0.1.31 after 3 attempts each with waiting timeout of 0.5 seconds

```

### Detailed context

 I verified that I can reach the board by the `ping`.

I didn't update or change the fw nor used the arms.

filippoborgogni, 2025-04-09:
Hi @valegagge, this morning I checked this issue and I were able to observe that the EB31 board started by default in eUpdater. 
I set the board to start in the eApplication via the `FirmwareUpdater` with the command `Set Def Boot eApplication` and afterwards I ran `yarprobotinterface` a couple of times and the problem did not occur again. 
Give us a feedback! 🚀

cc @AntonioConsilvio @LoreMoretti

AntonioConsilvio, 2025-05-05:
Hello! Since this problem was no longer reported, I think @filippoborgogni managed to solve it. Good job! ✅

closed 2025-05-05 by AntonioConsilvio