robotology/mesh-support #2112: Several fingers do not close fully or not functional

S-Dafarra, 2025-04-09:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

Here the list of issues:
- the right index OC closes up to 45 deg (the motor is at ~-26000) and it does not open
- the right thumb OC closes at 60deg (the motor is at -30000)
- the right index adduction does not move (the encoder reads something, about 10 deg)
- the left ring OC closes at 54deg (the motor is -20000)
- the left index adduction does not move and the encoder is stuck at 0

### Detailed context

We noticed the fingers having some issues after some teleoperaation test

cc @fbiggi @fiorisi @lawproto @andreaderito

PasMarra, 2025-04-09:
Also:
- right: middle, ring_pinky
- left: index_oc

S-Dafarra, 2025-04-22:
> > Also:
> > - right: middle, ring_pinky
> 
> The middle finger stopped working during a teleoperation session late Friday. Just posting for documentation

This is alarming. It seems that those fingers lasted about a single day as they got a complete overhaul on Thursday

@Fabrizio69 @fiorisi @maggia80

AntonioConsilvio, 2025-04-22:
Hi @S-Dafarra @mebbaid. We performed a complete checkup of the robot's mechanics between Wednesday and Thursday last week, (as quoted by @S-Dafarra), and now all fingers perform the full mechanical range of motion.

However, the tendon of the right middle finger broke (as quoted by @mebbaid), but for now we can relax, the reason for the rupture is that the tendon slipped out of the motor-side crimp. In fact, the crimp is tightened by hand, so it can happen that the crimp is not tightened sufficiently; we can consider this a human error.

However, we have already taken steps to prevent these occasions from happening again, by buying machine crimped tendons which will be assembled on the robot in the near future!

However, @fbiggi replaced the tendon and now the finger is working properly! Closing! ✅

closed 2025-04-22 by AntonioConsilvio