robotology/mesh-support #2116: left upper leg joints go in hardware fault when moving hip yaw to -38 deg

LoreMoretti, 2025-04-15:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

When driving the `left hip yaw` to -38[deg] through the `YarpMotorGui` all the left upper leg joints go in hardware fault.

### Detailed context

It is not possible to put back the joints in `idle` mode.

The fault message is `no fault detected`

[image]

filippoborgogni, 2025-05-14:
Hi @LoreMoretti,
today with @edersonferrante I worked to solve this issue. 
I could not get the problem to reoccur, despite the fact that we positioned all the joints according to the situation described. I checked the wiring (CAN connections) connecting the ems4 board (label EB8) with the two 2FOC boards (label 8B0 and 8B1). I could see that the cable connecting the two 2FOC boards (label 8N1) was damaged (as can be seen in the photo below) due to the fact that it was pinched when the joint of the `left hip yaw` was moving inwards (and therefore towards the degrees shown in the issue). 

[image]

I then made a new, slightly longer (about 1/2 cm) wire so that it could pass further out so as not to be pinched by the mechanics during movement.
As can be seen in the video below, the robot works properly and the joint is able to move throughout the `range of motion`.

https://github.com/user-attachments/assets/138978a7-4737-48df-abba-cd6d0cbef47d

Please let us know if the problem recurs, so that we can close this issue! 🚀

cc @AntonioConsilvio @Fabrizio69

LoreMoretti, 2025-05-27:
We can close this then. Thank you for investigating!