robotology/mesh-support #2117: Right Shoulder Tendon Failure

Ep3896, 2025-04-16:
### Robot Name 🤖

iCubErzelli02 S/N:036

### Request/Failure description

Following today’s maintenance for issue #2098, the right shoulder tendon broke during the demonstration.

### Detailed context

cc @ddetommaso 

### Additional context

https://github.com/user-attachments/assets/50af68b4-dcd0-4e2d-a32d-3c042a6e7cdf

The attached video includes audio, and the sound of the tendon failure can be clearly heard at the moment it occurred.

### How does it affect you?

This problem is of high priority: it must be resolved to proceed with the experiments.

ddetommaso, 2025-04-22:
Hi @AntonioConsilvio 

could you please provide us a possible date for the intervention?
Thank you

edersonferrante, 2025-04-30:
Hi @ddetommaso , cable 6 Cod. 4763 and cable 1 Cod. 4758 were broken and cable 7 Cod. 2390 had come out of its housing but was intact.

[image]

- [x] Successful cable replacements 

[image]

https://github.com/user-attachments/assets/b4b86d07-3b93-4071-bba6-725d0ced04a7

https://github.com/user-attachments/assets/43d17142-88ff-42c2-a017-ee9a208e07a3

cc @AntonioConsilvio @Fabrizio69