robotology/mesh-support #2119: Right shoulder roll does not calibrate

LoreMoretti, 2025-04-29:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

When starting the robot, the right shoulder roll does not calibrate.

I have tried twice.

### Detailed context

Once the whole calibrati in procedure is completed, the robot ends up in the following configuration:

[image]

with the right shoulder roll being in idle

LoreMoretti, 2025-04-29:
Here the error message I see:

[ERROR] from BOARD 10.0.1.1 (right_arm-eb1-j0_1) time=50s 147m 625u :  HW: strain values saturation. 0 is the channel involved. In the last second, the lower saturation counts is 0 and the upper one is 6
[ERROR] from BOARD 10.0.1.1 (right_arm-eb1-j0_1) time=50s 147m 726u :  HW: strain values saturation. 4 is the channel involved. In the last second, the lower saturation counts is 20 and the upper one is 0

MSECode, 2025-05-08:
Hi @valegagge and @LoreMoretti,
yesterday I checked the problem with the AMO encoder of that joint. In my case it calibrated but during the parking I needed to fault the joint since it started to going towards the hw limit without being correctly controlled (as I suppose it happened to you).
Checking the logs I've seen that we have a fail in the reading routine of the amo encoder board, specifically in the part that is related to the startup of the reading.
Having errors on that bits basically means that the AMO encoder board is not able to track the reading of the magnetic disk. Following tests that we have done on a bench setup this is due to 2 possible reasons (considered what we know up to now):
- position of the amo board is too far from the magnetic disk
- data cables that sometimes do not make contact 

The error I got in the logs is the following:
```
[WARNING] from BOARD 10.0.1.2 (left_arm-eb2-j0_1) time=209s 729m 21u :  HW - amo encoder status0 error: poor-level or clipping in the signal of master and/or nonius track. (Joint= (NIB=0) Number of poor-level or clipping errors in 10 seconds for nonius track is: 10 and for master track is: 10)
[WARNING] from BOARD 10.0.1.2 (left_arm-eb2-j0_1) time=209s 729m 136u :  HW - amo encoder status1 error: errors on reading on master and/or nonius track. (Joint= (NIB=0) Number of error in 10 seconds as internal crc errors: 0, i2c communication errors: 0, nonius track period consistency errors: 0, eccessive input signal frequency errors: 1)
```

Errors related to `status 1` are less blocking since they are mainly related to reading errors correlated to electromagnetic noise (similar to crc related errors).
Instead errors related to `status 0` are basically blocking since you cannot have a valid reading from the encoder.

[image]

Thus, as a solution I'll propose this:
1. check the cabling
2. check if AMO encoder disk is damaged 
3. check the position of the board (in the documentation is defined the precise distance it should have from the disk otherwise ask mechanics)
4. recalibrate the amo using its own tool
5. re-check joint calibration

filippoborgogni, 2025-05-14:
Hi @MSECode @LoreMoretti, 
today with @edersonferrante I worked to resolve this issue. 
I initially checked the cable (label RAS1) that connect the AMO sensor of the joint to the ems4 board (label EB1) and it was intact. Then I saw that the AMO sensor was not disabled in the `hardware/motorControl/right_arm-eb1-j0_1-mc_service.xml` file, and this is strange, as when a sensor does not read correctly or if it is damaged, it is immediately disabled in this file: this means that despite thinking that the sensor was reading incorrectly, the robot continued to start and calibrate the joint of the `right shoulder roll` using the values taken from this sensor. 
However, I carried out some motion tests to get a better understanding of the situation and to see if the AMO reading was getting out of whack during use. As you can see in the video below, the joint continued to function correctly throughout its `range of motion`. 

https://github.com/user-attachments/assets/872121e3-2d6b-412d-970b-4abbc51bca90

Please let us know if this situation recurs so that we can investigate this problem. 

cc @AntonioConsilvio @Fabrizio69

LoreMoretti, 2025-05-27:
Ok, thanks! For me we can close then.

AntonioConsilvio, 2025-07-09:
Thank you @LoreMoretti, Closing! ✅

closed 2025-07-09 by AntonioConsilvio