robotology/mesh-support #2121: Neck yaw issue

ddetommaso, 2025-05-02:
### Robot Name 🤖

iCubErzelli01 S/N:032

### Request/Failure description

The neck has some problem moving towards the joints limits (yaw).

### Detailed context

The issue is systematic and occurs when commanding to move the neck yaw from yarpmotorgui towards the joint limits. The issue occurs moving the neck in both direction and at different joint speeds (tested between 10 and 20).

[image]

### Additional context

[log_icub-head_yarprobotinterface_1493.txt](https://github.com/user-attachments/files/20015000/log_icub-head_yarprobotinterface_1493.txt)

### How does it affect you?

CRITICAL! We need to start data collection soon

ddetommaso, 2025-05-06:
Hello guys,

I would please ask you for a tentative date for the intervention.
Thank you

cc @AgnieszkaWykowska

AntonioConsilvio, 2025-05-07:
Hi @ddetommaso, We can intervene tomorrow morning!

filippoborgogni, 2025-05-08:
Hi!
Today @edersonferrante and I worked on solving this issue.
We initially disassembled the head and tightened the screws on the mechanical assembly of the `neck yaw` joint.

[image]

We then moved the position of the neck sheath with clamps on the mechanics as tightening the screws did not solve the problem (you can see the position in the photo). 

[image]

We then found some problems in the `range of motion` of the `neck yaw` when the `neck pitch` joint was backwards; we could see that this problem only occurred if the `neck pitch` joint was more than ~13/14 degrees: unfortunately this problem cannot be solved since it is structural, but it can be worked around by modifying the position of the `neck pitch` to obtain the required placement for the experiments. 
In the video below you can see, using the `yarpmotorgui`, the fact that the `neck yaw` struggles to reach the left limit with the `neck pitch` at the hardware limit.

https://github.com/user-attachments/assets/b1431e2e-7b46-4c0d-a423-1b0e1fa1e915

In the video below you can see how the problem is solved (in an optimal situation with the other neck joints at 0 or without the `neck pitch` being over 14 degrees) and the `neck yaw` joint is able to move perfectly in the whole `range of motion`. 🚀

https://github.com/user-attachments/assets/8df49fe3-04fb-4fb5-8b61-17877de0bc12

cc @AntonioConsilvio @Fabrizio69

ddetommaso, 2025-05-08:
Hi @filippoborgogni 

thank you for the detailed report.
I would please ask you to wait closing this issue because we will need to run a few more tests to see if this workaround can be applied in our scenario. 

Thank you,

ddetommaso, 2025-05-20:
Hi @AntonioConsilvio,

we ran the experiment with the workaround for some days and, even if the issue was mitigated, the problem is still there and it seems getting worse. Additionally, we ran the exact same experiment on another robot (iCubErzelli02) and the issue with the neck wasn't there.

Please we need your support to definitively fixed the problem.

cc @Ep3896 @AgnieszkaWykowska