robotology/mesh-support #2122: Right arm ring finger remains bent

Ep3896, 2025-05-02:
### Robot Name 🤖

iCubErzelli02 S/N:036

### Request/Failure description

the right-arm ring finger remains bent

### Detailed context

 The right finger remains bent at the end of the calibration process and does not align with the right pinky. 

### Additional context

[image]

### How does it affect you?

critical problem: the robot is unable to perform grasping and data collection needs to start as soon as possible

ddetommaso, 2025-05-06:
Hello guys,

I would please ask you for a tentative date for the intervention.
We will need the robot fully working for an institutional demo we have in a couple of weeks. So please proceed scheduling further checks if needed.

Thank you

cc @AgnieszkaWykowska

filippoborgogni, 2025-05-08:
Hi!
Today @edersonferrante and I have been working to resolve this issue.
After trying to give the tendons the right tension and observing that the problem persisted, we changed the calibration values in the configuration file `right_arm-calib.xml` concerning the fingers that remained closed (middle and pinky) ([link to PR already approved](https://github.com/robotology/robots-configuration/pull/729)). 
Now the robot works properly and, as can be seen in the video below, middle and pinky fingers close and open in the correct way. 

https://github.com/user-attachments/assets/8f574275-8e98-481d-9e45-eeeffba8b4e9

Let us know when you have aligned the `robots-configuration` to the PR so we can close this issue! 🚀

cc @AntonioConsilvio @Fabrizio69

AntonioConsilvio, 2025-07-16:
Hi! I think the configuration files have been aligned at this time, I will continue with the issue closure. ✅

closed 2025-07-16 by AntonioConsilvio