robotology/mesh-support #2126: iCubGenova07 S/N:040 – The cable of the left sholder got broken

omareldardear, 2025-05-15:
### Robot Name 🤖

iCubGenova07 S/N:040

### Request/Failure description

The cable of the left arm broke.  It is required to be replaced with a new cable

[image]

### Detailed context

We were executing a motion using the cptService, which is usually executed without problems.

I also include @reafrancesco  to follow up 

Thanks

AntonioConsilvio, 2025-05-15:
HI! We only have availability tomorrow morning. Is that OK with you?

cc @omareldardear @reafrancesco

omareldardear, 2025-05-15:
@AntonioConsilvio yes. This will be great. 
Thank you. 

I also added this video, which was captured just before it broke. the arm motion was't smooth and we realized it after wards 

https://istitutoitalianotecnologia-my.sharepoint.com/:v:/g/personal/omar_eldardeer_iit_it/EZ6j1eUJCWdAuSv9Z6SVUv8BXZmroP0qYynRvGFDblk2bQ?nav=eyJyZWZlcnJhbEluZm8iOnsicmVmZXJyYWxBcHAiOiJPbmVEcml2ZUZvckJ1c2luZXNzIiwicmVmZXJyYWxBcHBQbGF0Zm9ybSI6IldlYiIsInJlZmVycmFsTW9kZSI6InZpZXciLCJyZWZlcnJhbFZpZXciOiJNeUZpbGVzTGlua0NvcHkifX0&e=BPmrL9

AntonioConsilvio, 2025-05-15:
Great!

> [@AntonioConsilvio](https://github.com/AntonioConsilvio) yes. This will be great.

 Thank you for the video!

edersonferrante, 2025-05-16:
Hi @omareldardear, me and @AntonioConsilvio fixed the left shoulder.
There was cable 8 Cod. 3492, cable 4 Cod. 4761 that were broken and we replaced them instead there was cable 7 Cod. 2390 that had come out of its housing.

[image]

- [x] Successful intervention.

https://github.com/user-attachments/assets/ff691163-28ca-41e0-9f4f-23818cefb631

cc @Fabrizio69

closed 2025-05-16 by edersonferrante

omareldardear, 2025-05-19:
Many Thanks :)