robotology/mesh-support #2131: Right hand thumb position error

Ep3896, 2025-05-19:
### Robot Name 🤖

iCubErzelli02 S/N:036

### Request/Failure description

The right hand thumb is in a erroneous position 

### Detailed context

cc  @ddetommaso 

### Additional context

[image]

### How does it affect you?

critical problem: an important demo is scheduled for next week

AntonioConsilvio, 2025-05-20:
Hi @Ep3896! I have a few questions to better understand the situation:

- Should the thumbs of both hands (especially the thumb oppose joints) be in the same position? 

- What would be the desired degrees that the joints should reach?

- Can we have a picture of the yarpmotorgui of the 3 thumbs joints (both left and right hand)?

- Is the thumb oppose moving properly?

Thank you in advance for the information provided!

Ep3896, 2025-05-21:
Hi @AntonioConsilvio ! 

**Q1**:  Should the thumbs of both hands (especially the thumb oppose joints) be in the same position?

 Yes! thumbs have to be in the same position. Unfortunately they are visibly in different positions even though , using yarpmotorgui, from a software side it shows erroneously that they are in the same position.

**Q2**:  What would be the desired degrees that the joints should reach?

These joint should have the same values displayed on yarpmotorgui and so with the same values of the joints in the opposite hand.

**Q3**: Can we have a picture of the yarpmotorgui of the 3 thumbs joints (both left and right hand)?

[image]

[image]

**Q4**: Is the thumb oppose moving properly? 

As far we have seen, there are no detected problems on the thumb oppose movement. Can it be a calibration problem?

AntonioConsilvio, 2025-05-27:
Hi @Ep3896, thanks for the fundamental info provided, now the situation is clear.
The problem is almost certainly a calibration problem, which you should be able to solve easily. 
We will be at Erzelli tomorrow to work on this issue:

- https://github.com/robotology/icub-tech-support/issues/2121

If enough time is left over, we will try to solve this one as well.

gioelemigno, 2025-06-16:
Hi guys,

We need to start an experiment very soon using this robot, could you please provide an update about this issue?

Please provide an update also to [this other issue](https://github.com/robotology/icub-tech-support/issues/2121).

Thank you

cc @ddetommaso

AntonioConsilvio, 2025-06-17:
Hi! Certainly the thumb oppose of the right hand (joint 8) needs to be recalibrated, I leave you the guide to do this.

- https://icub-tech-iit.github.io/documentation/icub_robot_calibration/icub-robot-calibration-v2.x/#hands-calibration

If you are unable to do this, we are available to answer any questions you may have or, if necessary, to organise on-site support.
However, I would first suggest trying the recalibration procedure to see if you succeed!

cc @gioelemigno @ddetommaso

AntonioConsilvio, 2025-07-16:
Hi @gioelemigno @ddetommaso, do you have any update on this issue?

closed 2025-09-25 by AntonioConsilvio