robotology/mesh-support #2146: Robot calibration fails

andrearosasco, 2025-06-05:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

When I try to run yarprobotinterface on the torso to calibrate the robot it fails 

### Detailed context

The error message is shown in the attached screenshot.

[image]

andrearosasco, 2025-06-10:
Update: the robot dont calibrate the left leg anymore, and the command 

yarprobotinterface --disable_tags "(disable_right_arm, disable_right_hand, disable_left_leg, disable_right_leg)"

, that is just one arm, cant calibrate the robot correctly anymore,

[image]

filippoborgogni, 2025-06-16:
Hi @andrearosasco ,
last week I worked to solve this issue and [this one](https://github.com/robotology/icub-tech-support/issues/2143).
I noticed that the cable (label 22N0) that connects the mc4-plus-cshape board (label EB22) to the mtb4 board (label 22N1) was damaged. I crimped it and now it works properly.
Regarding the issue about the left arm, I fixed the cable (label 4N2) that connect the ems4 board (label EB4) to the mtb4 board (label 4B2) that was broken. 
Now the robot works properly, closing!🚀

cc @AntonioConsilvio