robotology/mesh-support #2147: Unable to detect the ergocub-torso

GiulioRomualdi, 2025-06-06:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

We tried to switch on the robot and connected the Ethernet cable at the back. From the laptop, we can ping and access `10.0.0.3` (ergocub-head), but we cannot reach `10.0.0.2` (ergocub-torso), either from the head or the laptop.

### Detailed context

. 

### How does it affect you?

We cannot start the robot

davidelasagna, 2025-06-09:
The  `ergocub-torso` cannot boot since it seems that the EMMC ( the internal soldered disk) of the comexpress is damaged, I've tried to reinstall the OS but the disk cannot be mounted nor read and written, even trying to delete the partitions does not work, I've also tried to force a write of all zeros on the disk but it did not have any effect. 
I think the only solution available is to replace the comexpress board.

@aazocar7 @AntonioConsilvio @Fabrizio69 @maggia80

filippoborgogni, 2025-06-16:
Hi @GiulioRomualdi ,
@aazocar7 replaced the COM-EXPRESS.
With the help of @davidelasagna, we programmed and set the network of the new COM-EXPRESS.
Now the robot works properly, closing!🚀

cc @AntonioConsilvio

closed 2025-06-16 by filippoborgogni