robotology/mesh-support #2148: Right pinky and right hand adduction not moving

martinaxgloria, 2025-06-09:
### Robot Name 🤖

iCubGenova11 S/N:050

### Request/Failure description

`r_pinky` and `r_hand` don't calibrate.

### Detailed context

`r_pinky` doesn't calibrate, probably due to a broken tendon. For the `r_hand` joint, instead, remains in idle mode and the motor runs loose.

AntonioConsilvio, 2025-06-27:
Hi @martinaxgloria, as the repair takes too long to be carried out in the short term for now I will remove the joint from calibration to avoid more damage.

- https://github.com/robotology/robots-configuration/pull/744

AntonioConsilvio, 2025-10-02:
HI! @edersonferrante fixed also the adduction tendon.

Closing!

closed 2025-10-02 by AntonioConsilvio