robotology/mesh-support #2158: Neck pitch faults, but not on yarpmanager

steb6, 2025-06-18:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

When there is a sudden movement, the neck pitch should go into hardware fault, but sometimes it happens that it remains in a idle state with the yarpmotorgui that doesn't allow to move it or calibrate it (it just doesn't move the joint). If I shut down the motors and calibrate the robot, it works again for a while

### Detailed context

look above

pattacini, 2025-06-18:
>it happens that it remains in a idle state with the yarpmotorgui that doesn't allow to move it or calibrate it (it just doesn't move the joint). 

When you say idle state, do you mean that the pitch is colored in yellow `#FFFF00` within the yarpmotorgui?

pattacini, 2025-06-18:
Ok, thanks heaps! Now it's clearer to me.

Nicogene, 2025-06-19:
@AntonioConsilvio what is the version of the firmware running on the amc/amcbldc?

Because @MSECode worked on a very similar issue:
- https://github.com/robotology/icub-firmware-build/pull/209

pattacini, 2025-06-25:
/remind July 04 check the status and close if possible.

steb6, 2025-06-25:
I just tested it with teleoperation and it worked perfectly, even under a bit of stress. I think we can close this, thank you!

S-Dafarra, 2025-06-30:
It happend again today while preparing for a demo. Interestingly, the robot has been faulted earlier, and then it seemed to remain faulted even if it was green from the motorgui

MSECode, 2025-06-30:
Just a question regarding that, did the calibration end fine after the joint went in fault at the restart of the `yarprobotinterface`? Because if we did not have a correct calibration where the fault should forced to be cleaned we might take a deeper look to this behavior?

pattacini, 2025-06-30:
cc @mfussi66

Nicogene, 2025-07-02:
@martinaxgloria and @MSECode tested the neck pitch after pressing the fault, and everything seemed to work as expected:
- https://github.com/icub-tech-iit/study-icub-headedge/issues/252#issuecomment-3023673179

We can close it, keeping an eye on this, in case it happens again, it may be some race condition happening

closed 2025-07-02 by Nicogene