robotology/mesh-support #2159: Motors starts with panic(red) lights on in all parts of the body

MrHuangWenjie, 2025-06-23:
### Robot Name 🤖

iCubManchester01 S/N:011

### Request/Failure description

We are setting up the robot after maintenance. The power supply was correctly set, but the panic (red) lights are always on after the motors are turned on. Please check the attached images for specifics.

[image]
[image]
[image]

### Detailed context

The power supply was set with 13 V for the upper deck and 37 V for the bottom deck (motor power). We followed the steps, turning on upper deck first, then bottom deck, then CPU green switch and finally motor green switch. After we turned on the motor switch, all parts of the robot body show red lights.

We might need some technical support by video supervision to set up our iCub again in our lab. 

Many thanks,
CoRoLab, University of Manchester

AntonioConsilvio, 2025-06-30:
Hi @MrHuangWenjie, sorry for the delay in replying, but we had to work on some high-priority tasks. We will now try to provide you with maximum support.

Unfortunately, it is difficult for us to determine the exact meaning of the LEDs on those boards as they are an old version. 
However, I would suggest you run the `yarporobotinterface` and provide us with the log so we can get more information.

Any errors that might affect all the boards would usually to concern the fault button (red button), which might have some connection problem and send all the boards into fault, but this is just a speculation.

MrHuangWenjie, 2025-06-30:
Hi Antonio,

Thank you for the reply. I was just trying to confirm that the red lights do not mean something emergent. 

The good news was that I managed to connect to icub-head. This seems like that the red lights are not something stopping us from connecting to iCub. The issues now I think are:

1. As for the $yarporbotinterface$ you mentioned, I am sorry that I am not sure which command or application you are referring to. Sorry, I am quite new to iCub. I appreciate a lot if you can specify it more.

2. In addition, there is nothing in the home folder of icub-head but a config file. Do I need to install everything from beginning for it? If yes, which instruction should I follow please?

Sorry for that if my questions are too naive.

Many thanks,
Wenjie

MrHuangWenjie, 2025-07-01:
Hi Antonio, 

I was also expecting that the robot is configured already. What I got when I run yarp run on icub-head was ‘-bash: yarp: command bot found’

MrHuangWenjie, 2025-07-01:
Hi Antonio,

Good news, I managed to solve the problem. It seems like an environment issue.

I am now with iCub functioning well.

Many thanks for your help.

Wenjie

AntonioConsilvio, 2025-07-02:
I'm glad to hear it.

I hope you will have a good experience with iCub.

If there are further problems in the future, feel free to open new issues.

Closing! ✅

closed 2025-07-02 by AntonioConsilvio