robotology/mesh-support #2167: Broken thumb in the left hand

CarmelaCalabrese, 2025-07-01:
### Robot Name 🤖

R1 S/N:003

### Request/Failure description

Yesterday, during a movement, the robot began to fall forward, so the fault button was pressed. While trying to return it to the home position, it fell down. We tried to catch it, but it hit the floor with its hand and the thumb broke.

### Detailed context

See above.

filippoborgogni, 2025-08-04:
Hi @CarmelaCalabrese @lornat75 ,
we are currently waiting for the parts to be produced by the workshop. 
@Mick3Lozzo has already created the order and, as soon as the parts arrive, we will be able to proceed with the replacement!
Thanks for your patience! 

cc @pattacini @Fabrizio69

closed 2025-08-06 by filippoborgogni

pattacini, 2025-08-07:
> thank you [@filippoborgogni](https://github.com/filippoborgogni) can you link the other issue you opened so that we can follow the debugging of the thumb add joint?

Hi @lornat75 

It's:
- #2211