robotology/mesh-support #2172: Left thumb adduction not working

S-Dafarra, 2025-07-04:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

The left thumb adduction does not seem to work

### Detailed context

The encoder reads continuously zero and the PID controller diverges

S-Dafarra, 2025-07-07:
Hi @AntonioConsilvio @filippoborgogni it is possible it was the right instead of the left, I started having some doubts. Apologies

AntonioConsilvio, 2025-07-07:
Hi @S-Dafarra, no problem! We will check both, thank you for the info!

S-Dafarra, 2025-07-07:
I checked on the robot and they both move now. I asked also @carloscp3009 and he remembers it was the right (and I wrote "right" also in my notes). Maybe a connector is a bit faulty? 🤔

AntonioConsilvio, 2025-07-09:
Hi @S-Dafarra, today I checked the status of the thumb adduction joints. The problem was on the right hand. 

As you can see, the cable that connect the FAP (encoder) to the MMA board was quite damaged as it was stuck in the mechanic.

[image]

I replaced it and it is working properly now! Closing! 🚀

cc @carloscp3009

AntonioConsilvio, 2025-07-16:
Hello @S-Dafarra! I found the cable mentioned in the previous comment damaged again. 
It was shorting out on the mechanics and thus shorting out the MMA part that controls the encoders of the adducts.

This therefore generates the warning mentioned in this issue:

- https://github.com/robotology/icub-tech-support/issues/2180

I have replaced the cable and it is now working properly! 

Closing! 🚀