robotology/mesh-support #2200: Hip pitch harmonic drive screws are loose

FabioBergonti, 2025-07-25:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

The right hip pitch has backlash and does not work as expected

### Detailed context

It seems that the screws of the harmonic drive of the right hip pitch are loose

AntonioConsilvio, 2025-07-25:
Hello! Me and @fbiggi removed the right leg apart to tighten the screws holding the gearbox.

[image]

https://github.com/user-attachments/assets/b0e74a50-450e-4f51-b8ad-02c714c7b714

We also decided to replace them with high-strength (black) screws so that the problem shouldn't recur in the future.

[image]

Since we had to disassembled everything, we decided to perform the same operation on the left leg.

During testing, we realised that the ankle motor was unscrewing. 

https://github.com/user-attachments/assets/b7f35688-bf17-4bf7-aa16-3397c16518b0

So I tightened it and now the robot works properly now! ✅

Mick3Lozzo, 2025-07-29:
[Snapshot updated](https://github.com/icub-tech-iit/robot-snapshots/commit/d27bb8161c6ca671bc129ac6e7a4b3703fc2d6ae) @fbiggi