robotology/mesh-support #2210: Right knee is not able to calibrate because the motor encoder is not working fine

fils99, 2025-08-01:
### Robot Name 🤖

ergoCub 1.1 S/N:001

### Request/Failure description

When launching the `yarprobointerface`, the robot is not able to perform the calibration of the right knee.

### Detailed context

The problem is probably due to the fact that the motor encoder is not working fine. As I will show in the image below, it always measures 0.0 deg.

cc @S-Dafarra 

### Additional context

[image]

fils99, 2025-08-29:
Great, thanks!

AntonioConsilvio, 2025-09-04:
Thank you @aazocar7, good job!