robotology/mesh-support #2219: l_thumb_add in hw fault after calibration

PasMarra, 2025-08-21:
### Robot Name 🤖

ergoCub 1.3 S/N:002

### Request/Failure description

as per title, `l_thumb_add` in hw fault after calibration

### Detailed context

[image]

PasMarra, 2025-09-02:
I've just calibrated the robot and the problem is back again:

[image]

AntonioConsilvio, 2025-09-04:
HI! the calibration files was not updated properly.
This is the commit:

- https://github.com/icub-tech-iit/robots-configuration/commit/3bccd57e14a6ae3cd5423588db619d541f038620

Now the robot is updated and it is working properly. 

Closing! 🚀