robotology/mesh-support #2220: l_ring_pinky_oc does not cover the full RoM

PasMarra, 2025-08-21:
### Robot Name 🤖

ergoCub 1.0 S/N:000

### Request/Failure description

as per title, `l_ring_pinky_oc` motion is limited to `[0,35]` deg instead of `[0,90]` deg 

### Detailed context

[image]

lornat75, 2025-09-02:
Ciao @edersonferrante questo problema sta diventando urgente, riuscite ad intervenire? Grazie

edersonferrante, 2025-09-02:
Ciao @lornat75 , come concordato con @PasMarra interverremo domani mattina.

AntonioConsilvio, 2025-09-04:
Hi! @edersonferrante replaced the tendon and it is working properly now! 

https://github.com/user-attachments/assets/31b97cb2-df7c-4556-9593-526e079fc78c

There is only one small problem with the ring finger. The finger has some backlash caused by wear on the plastic. This backlash causes a slight difference between the closure of the ring finger and the little finger.

However, this would require replacement of the phalanx.

In any case, the position is measured on the little finger, so the entire range of motion is covered because the little finger is working properly.

I am proceeding with closing the issue. If this minor malfunction causes problems, please feel free to open another issue regarding it.

closed 2025-09-04 by AntonioConsilvio